haomo-ai/OverlapTransformer #1: gt_valid_folder and cov_files

smalltheater, 2022-04-05:
Hello authors,

thank you for sharing your work!

Is the files in gt_valid_folder generated from OverlapNet demo? And How to get the cov_file from suma++?

I’m looking forward to hearing from you :)

BIT-MJY, 2022-04-05:
Hi @smalltheater ,

Thanks for using our code! 

You are right. gt_valid_folder can be generated from OverlapNet [demo4](https://github.com/PRBonn/OverlapNet/blob/master/demo/demo4_gen_gt_files.py), where you should use sequence 02 to compute overlaps for the validation. Each row in the ground truth files is a triplet (idx_1, idx_2, overlap_between_1_2). 

As to the cov_file from Suma++, @Chen-Xieyuanli  could you please give more details of the covariance generation here?

Chen-Xieyuanli, 2022-04-05:
@smalltheater For more details on calculating the covariance, you could find it in our previous [paper](https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/chen2021auro.pdf). We will also release the cov_file used in our OverlapTransformer after the paper gets accepted.

closed 2023-05-26 by Chen-Xieyuanli