haomo-ai/OverlapTransformer #2: Will a coarse rotation guess will be provided after a place recog?

Jarrome, 2022-06-10:
Hi, Ma and Chen,

Really nice contribution! Your released code is easy to run and queries really fast!

Here is one request: 

From my understanding, this model provides a similarity score. Which is good enough to find old visit. 
But I didn't find a coarse rotation guess which is also interested in the application. Just as scancontext did.

I would be really happy if you could give me more information about the possibility of get a coarse yaw from your model. :D

BIT-MJY, 2022-06-10:
Hello @Jarrome ,

Thanks for your nice comment!

OT now can only provide a similarity score for retrieval because the devised end-to-end architecture is used to generate yaw-angle-invariant descriptors. Fortunately,  the feature volumes generated by Transformer module are yaw-angle-equivariant (see Section III-B in our [paper](https://ieeexplore.ieee.org/document/9785497)), which can be used to regress out the yaw angle following the correlation head of [OverlapNet](https://github.com/PRBonn/OverlapNet). Thus, a simple siamese architecture is needed for estimating the yaw angle of the two input scans, leading to an increasing runtime. 

It would be really cool If yaw estimation is achieved by a new siamese network based on OT :)

Jarrome, 2022-06-10:
> Fortunately, the enhanced feature volumes generated by Transformer module are yaw-angle-equivariant (see Section III-B in our [paper](https://ieeexplore.ieee.org/document/9785497)), which can be used to regress out the yaw angle following the correlation head of [OverlapNet](https://github.com/PRBonn/OverlapNet).

That would be great. I would close this question as Junyi gives a good direction for the coarse estimation to use like scancontext. If would be interesting if an following extension is approaching.

closed 2022-06-10 by Jarrome