haomo-ai/OverlapTransformer #5: 测试集ground truth

Zhilong-Tang, 2022-07-28:
你好，关于你们代码中给的00sequence的ground truth文件内容好像有点问题，只有关于最后几帧的数据，另外请问你们是如何得到loop_gt_seq00_0.3overlap_inactive.npz. 期待你们的回复，谢谢！

BIT-MJY, 2022-07-28:
您好 @Zhilong-Tang ，

感谢使用我们的代码！

ground truth文件在上传前已经经过检查无误。由于KITTI00具备回环的地点并不是连续的，因此某些帧并没有回环的真值帧序号，所对应的列表为空。您可以对ground truth文件中的数据进行循环，可以检查到所有有真值的帧序号，并非仅有最后几帧具备真值。

closed 2022-08-01 by BIT-MJY

Zhilong-Tang, 2022-08-10:
谢谢您的回复，但是我还是有一些问题不太理解。
  我尝试去对02序列进行测试，因此我想要生成一个类似loop_gt_seq00_0.3overlap_inactive.npz的文件，但是在运行Demo4以后，我得到了ground_truth_overlap_yaw.npz, train_set.npz 和validation_set.npz文件，在读取这些文件的数据之后，发现和loop_gt_seq00_0.3overlap_inactive.npz差距很大，不知道该如何利用这些文件来生成loop_gt_seq00_0.3overlap_inactive.npz这样的ground_truth文件。
另外想请教一下，在我阅读pointnetvlad系列代码时，发现他们都ground_truth文件的生成都是基于点的gps坐标（并不需要读取点云文件)，而您的OverlapNet中却需要读取calib和点云数据，请问这是基于什么原因呢。期待您的回答，谢谢

BIT-MJY, 2022-08-11:
> 不知道该如何利用这些文件来生成loop_gt_seq00_0.3overlap_inactive.npz这样的ground_truth文件

得到ground_truth_overlap_yaw.npz之后，可以进一步处理，将每一帧对应的overlap大于0.3的真值帧序号提取出来。

> 您的OverlapNet中却需要读取calib和点云数据，请问这是基于什么原因呢

两帧点云Overlap的计算需要根据6-dof位姿关系将一帧点云投影至另一帧点云坐标系中，因此需要位姿和点云相关文件。

BIT-MJY, 2022-10-30:
您好！我们已经上传了生成真值文件的脚本，具体请参考此[issue](https://github.com/PRBonn/OverlapNet/issues/35)。

NUDTherui, 2023-05-31:
> > 不知道该如何利用这些文件来生成loop_gt_seq00_0.3overlap_inactive.npz这样的ground_truth文件
> 
> 得到ground_truth_overlap_yaw.npz之后，可以进一步处理，将每一帧对应的重叠大于0.3的真值帧序号提取出来。
> 
> > 您的OverlapNet中却需要读取calib和点云数据，请问这是基于什么原因呢
> 
> 两帧点云重叠的计算需要根据6-dof位姿关系将一帧点云投影至另一帧点云坐标系中，因此需要位姿和点云相关文件。计算重叠的公式可见我们的[论文Section III-A](https://arxiv.org/pdf/2203.03397.pdf)。

> 为什么要筛选大于0.3的重叠，而Seqot里不需要筛选只需要位姿小于15即可

ynma-hanvo, 2024-05-23:
> > 不知道该如何利用这些文件来生成loop_gt_seq00_0.3overlap_inactive.npz这样的ground_truth文件
> 
> 得到ground_truth_overlap_yaw.npz之后，可以进一步处理，将每一帧对应的overlap大于0.3的真值帧序号提取出来。
> 
> > 您的OverlapNet中却需要读取calib和点云数据，请问这是基于什么原因呢
> 
> 两帧点云overlap的计算需要根据6-dof位姿关系将一帧点云投影至另一帧点云坐标系中，因此需要位姿和点云相关文件。计算overlap的公式可见我们的[论文Section III-A](https://arxiv.org/pdf/2203.03397.pdf)。

hi,   I just  notice that overlap  is only used for generating trainset and threshold is 0.3.;     

in my case,  I need to test if two frame of pointcloud  is a loop,   how do i determine the threshold value between the two descriptor?