haomo-ai/OverlapTransformer #6: pretrained model doesn't produce results in the paper

mj735, 2022-07-31:
Hello, thanks for the great work!

I download your pretrained model on kitti, generate depth data using gen_depth_data.py, and run 

python test_kitti00_prepare.py
python test_kitti00_topN.py

but I get 0.883 for Top1 and 0.950 for Top1%, which are good enough but have a little gap to the 0.906 and 0.964 in the paper.

Could you please give me some suggestions to reproduce your results? Thanks!

BIT-MJY, 2022-08-01:
Hello @mj735 ,

Thanks for using our code!

Have you used the gen_depth_data.py provided in this repo? Or Have you changed it?

I think you should check the relevant setup carefully. The following screenshot of the topN recalls from our experiment is a reference:

[image]

If you still cannot get the results, I have some advise for improvement:
* increase the range in the `range_projection` function, such as 80 m.
* use .npy instead of .png.
* continue fine-tuning if necessary.

mj735, 2022-08-01:
Thanks for your reply. I will try it. 
Another question is that in the paper you said seq 03-10 are used for training. But in the config file there is no seq 10, only 03-09 (https://github.com/haomo-ai/OverlapTransformer/blob/85cbc1c3970c8339f91daccc7ac48875d3f5f167/config/config.yml#L19). Is it a bug or are there any reasons?

BIT-MJY, 2022-08-11:
It's a config param. You can change it according to my paper :)

closed 2022-08-21 by BIT-MJY