haomo-ai/OverlapTransformer #7: Is there a search space for evaluation on Haomo?

lqf211, 2022-08-11:
Hi authors,

Really nice work!

I have generated the results closed to the ones in your paper, but I want to know why there is no search space in the evaluation on Haomo dataset?

BIT-MJY, 2022-08-11:
Hi @lqf211,

Thanks for using our code!

Actually, we don't use any search space for retrieval while using either loop closure detection metrics or place recognition metrics. It's just used for better visualization in this repo. OverlapTransformer is a place recognition method instead of a loop closure detection method, so using a search space is not reasonable.

lqf211, 2022-08-21:
I found there is no search space while evaluating with either loop closure metric or place recognition metric. Thanks for your reply!

closed 2022-08-21 by lqf211