haomo-ai/OverlapTransformer #8: ford campus dataset

Bryan-ZhengRui, 2022-08-31:
Hello authors, thank you very much for sharing this project.
Due to the differences in the data formats of the ford campus dataset and the KITTI dataset, I can only process KITTI through the demo1 of Dr. Chen's OverlapNet codes. Therefore, I would like to know the main steps of processing the ford campus data  into range maps.

BIT-MJY, 2022-09-02:
Hello @Bryan-ZhengRui,

Thanks for your interest in our work!

You only need to modify the input param of the function `range_projection` in [gen_depth_data.py](https://github.com/haomo-ai/OverlapTransformer/blob/ad237fb0af21ac3f50c1e7a964a76c468efddf95/tools/utils/gen_depth_data.py#L9)  of this OT repo, or in [gen_depth_data.py](https://github.com/PRBonn/OverlapNet/blob/master/src/utils/gen_depth_data.py) in OverlapNet repo according to the LiDAR params used in Ford.

We are also considering releasing the pre-processing code for Ford.

Bryan-ZhengRui, 2022-09-03:
Thank you very much for your reply！I get it.

---- Replied Message ----
| From | Junyi ***@***.***> |
| Date | 09/03/2022 01:14 |
| To | ***@***.***> |
| Cc | Zheng ***@***.******@***.***> |
| Subject | Re: [haomo-ai/OverlapTransformer] ford campus dataset (Issue #8) |

Hello @Bryan-ZhengRui,

Thanks for your interest in our work!

You only need to modify the input param of the function range_projection in gen_depth_data.py of this OT repo, or in gen_depth_data.py in OverlapNet repo according to the LiDAR params used in Ford.

We are also considering releasing the pre-processing code for Ford.

BIT-MJY, 2022-10-25:
Hi @Bryan-ZhengRui, we have released all the necessary pre-processed files for Ford. You can find them [here](https://github.com/haomo-ai/OverlapTransformer#Datasets-used-by-OT).

Bryan-ZhengRui, 2022-10-26:
thank you for sharing!

closed 2022-10-30 by BIT-MJY