haomo-ai/OverlapTransformer #9: Deploying C++ OT model on ROS

SiqianLi, 2022-09-01:
Hello, thank you for your great work!

Have you tried deploying the C++ OT model on ROS? 
I am having a hard time running the OT model with ROS Kinetic in C++. The libtorch breaks the ROS when I try to include it in my package CMakelist.txt. But this "fast_ot.cpp" works in my environment compiled by cmake.

My Environment: 
Ubuntu 20.04, CUDA 11.1, Pytorch 1.10.0, Libtorch 1.8.0/1.9.0/1.10.0 (all I have tried, but with no success on ROS)

BIT-MJY, 2022-09-02:
Hello @SiqianLi,

Thanks for using our code!

Actually we have not implemented OT with ROS. If possible, could you provide the log related to your error here?

SiqianLi, 2022-09-03:
Here is my log at catkin_make:

[  8%] Building CXX object lidar_localization/CMakeFiles/data_pretreat_node.dir/src/mapping/loop_closing/loop_closing_flow.cpp.o
In file included from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader_options.h:4,
                 from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader/base.h:3,
                 from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader/stateful.h:3,
                 from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader.h:3,
                 from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data.h:3,
                 from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/all.h:8,
                 from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/torch.h:3,
                 from /home/ubuntu/learn/14nn_ws/src/lidar_localization/src/mapping/loop_closing/loop_closing.cpp:17:
/home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/types.h:39:16: error: conflicting declaration ‘constexpr const auto torch::kUInt8’
   39 | constexpr auto kUInt8 = at::kByte;
      |                ^~~~~~
In file included from /home/ubuntu/learn/14nn_ws/src/lidar_localization/third_party/libtorch/include/torch/script.h:3,
                 from /home/ubuntu/learn/14nn_ws/src/lidar_localization/src/mapping/loop_closing/loop_closing.cpp:16:
/home/ubuntu/learn/14nn_ws/src/lidar_localization/third_party/libtorch/include/torch/csrc/api/include/torch/types.h:39:16: note: previous declaration as ‘constexpr const c10::ScalarType torch::kUInt8’
   39 | constexpr auto kUInt8 = at::kByte;
      |                ^~~~~~

----------------------------------------------------------------------------
Following is my CMakelists.txt:

cmake_minimum_required(VERSION 2.8.3)
project(lidar_localization)

SET(CMAKE_BUILD_TYPE "Release")
#SET(CMAKE_BUILD_TYPE "Debug")
SET(CMAKE_CXX_FLAGS_RELEASE "$ENV{CXXFLAGS} -O3 -Wall")

add_compile_options(-std=c++14)
add_definitions(-std=c++14)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${TORCH_CXX_FLAGS}") 

set(CMAKE_PREFIX_PATH "/home/ubuntu/learn/14nn_ws/src/lidar_localization/third_party/libtorch/share/cmake/Torch;/opt/ros/noetic")

set(Torch_DIR ${PROJECT_SOURCE_DIR}/third_party/share/cmake/Torch)
find_package(Torch REQUIRED)
include_directories(${PROJECT_SOURCE_DIR}/third_party/libtorch/include/)
list(APPEND ALL_TARGET_LIBRARIES ${TORCH_LIBRARIES})

add_executable(loop_closing_node src/apps/loop_closing_node.cpp ${ALL_SRCS})
add_dependencies(loop_closing_node ${catkin_EXPORTED_TARGETS} ${PROJECT_NAME}_generate_messages_cpp) ${catkin_LIBRARIES} ${ALL_TARGET_LIBRARIES})
target_link_libraries(loop_closing_node PUBLIC ${catkin_LIBRARIES} ${ALL_TARGET_LIBRARIES})
-----------------------------------------------------------------------------
And attached is my running result of  ./fast_ot.
[image]

SiqianLi, 2022-09-03:
Here is my log at catkin_make:

```
[ 8%] Building CXX object lidar_localization/CMakeFiles/data_pretreat_node.dir/src/mapping/loop_closing/loop_closing_flow.cpp.o
In file included from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader_options.h:4,
from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader/base.h:3,
from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader/stateful.h:3,
from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data/dataloader.h:3,
from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/data.h:3,
from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/all.h:8,
from /home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/torch.h:3,
from /home/ubuntu/learn/14nn_ws/src/lidar_localization/src/mapping/loop_closing/loop_closing.cpp:17:
/home/ubuntu/Downloads/Tools/libtorch/include/torch/csrc/api/include/torch/types.h:39:16: error: conflicting declaration ‘constexpr const auto torch::kUInt8’
39 | constexpr auto kUInt8 = at::kByte;
| ^~~~~~
In file included from /home/ubuntu/learn/14nn_ws/src/lidar_localization/third_party/libtorch/include/torch/script.h:3,
from /home/ubuntu/learn/14nn_ws/src/lidar_localization/src/mapping/loop_closing/loop_closing.cpp:16:
/home/ubuntu/learn/14nn_ws/src/lidar_localization/third_party/libtorch/include/torch/csrc/api/include/torch/types.h:39:16: note: previous declaration as ‘constexpr const c10::ScalarType torch::kUInt8’
39 | constexpr auto kUInt8 = at::kByte;
| ^~~~~~
```

Following is my CMakelists.txt:

```
cmake_minimum_required(VERSION 2.8.3)
project(lidar_localization)

SET(CMAKE_BUILD_TYPE "Release")
#SET(CMAKE_BUILD_TYPE "Debug")
SET(CMAKE_CXX_FLAGS_RELEASE "$ENV{CXXFLAGS} -O3 -Wall")

add_compile_options(-std=c++14)
add_definitions(-std=c++14)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${TORCH_CXX_FLAGS}")

set(CMAKE_PREFIX_PATH "/home/ubuntu/learn/14nn_ws/src/lidar_localization/third_party/libtorch/share/cmake/Torch;/opt/ros/noetic")

set(Torch_DIR ${PROJECT_SOURCE_DIR}/third_party/share/cmake/Torch)
find_package(Torch REQUIRED)
include_directories(${PROJECT_SOURCE_DIR}/third_party/libtorch/include/)
list(APPEND ALL_TARGET_LIBRARIES ${TORCH_LIBRARIES})

add_executable(loop_closing_node src/apps/loop_closing_node.cpp ${ALL_SRCS})
add_dependencies(loop_closing_node ${catkin_EXPORTED_TARGETS} ${PROJECT_NAME}_generate_messages_cpp)
#target_link_libraries(loop_closing_node PUBLIC ${TORCH_LIBRARIES} ${catkin_LIBRARIES} ${ALL_TARGET_LIBRARIES})
target_link_libraries(loop_closing_node PUBLIC ${catkin_LIBRARIES} ${ALL_TARGET_LIBRARIES})
```

And attached is my running result of ./fast_ot which outputs the similar between frame 000000.bin and 000020.bin.
[image]

smalltheater, 2022-09-05:
I deployed this work on ROS, It might caused by wrong version of libtorch. You should not use **pre cxx 11** , it conflicted to your ROS c++ standard.

Chen-Xieyuanli, 2023-05-26:
It seems the issue has been solved. I will therefore close it. Please feel free to ask us to reopen it if needed.

closed 2023-05-26 by Chen-Xieyuanli