haomo-ai/OverlapTransformer #11: CUDA out of memory

Cxz-dev, 2022-10-24:
感谢你们的工作！
        我最近在RTX3090运行该代码的时候会报CUDA out of memory，但是论文中使用的是RTX3070，请问是哪里出问题或者我该怎么减小显存的使用？谢谢！

BIT-MJY, 2022-10-24:
@Cxz-dev 感谢使用我们的代码！

我们的显存为8G。您可以查看是否有其他程序占用了显存。

为保证代码的可用性我们已经将batchsize减小到较小的数值。目前使用6正样本6负样本，batchsize=13。您可以在[training_overlap_transformer_kitti.py](https://github.com/haomo-ai/OverlapTransformer/blob/master/train/training_overlap_transformer_kitti.py)文件中继续降低batchsize以保证不超出显存。

```
                """
                    the balance of positive samples and negative samples.
                    TODO: Update for better training results
                """
                use_pos_num = 6
                use_neg_num = 6
```

Cxz-dev, 2022-10-24:
好的谢谢！

closed 2022-10-24 by Cxz-dev