haomo-ai/OverlapTransformer #12: Fine Tuning and no. of epochs

surajiitd, 2022-11-09:
Dear Authors, Thanks for giving code for your novel method !!
I have few questions:
1. Can we fine-tune your kitti model on other datasets by making `self.resume` to True in this line. https://github.com/haomo-ai/OverlapTransformer/blob/da46c75eea877f46783f43b92481039e70065035/train/training_overlap_transformer_kitti.py#L60
2. You have set max epochs to 100 in your code. But when I am doing fine-tuning(self.resume=True), then the training is starting from 20th epoch. So have you trained your kitti model for 19 epochs ??  I can't find that in paper and README.
3. I am asking because the model is taking around 30mins for each epochs on kitti(whether fine-tuning or from scratch). So it's a bit long for training for 100 epochs.
4. you have skipped those anchors where neg_num(no. of negative samples) are 0 for the current anchor. In my dataset, I am getting pos_num as 0 for some anchors. Hope it's fine to also skip when `pos_num == 0`.
https://github.com/haomo-ai/OverlapTransformer/blob/da46c75eea877f46783f43b92481039e70065035/train/training_overlap_transformer_kitti.py#L151
@BIT-MJY @Chen-Xieyuanli

surajiitd, 2022-11-09:
And one more question, for how many epochs did you train you model in overlapnet_pytorch version.

BIT-MJY, 2022-11-14:
@surajiitd  Thanks for using our code!

1. Yes. You only need to prepare the processed training data correspondingly and modify the config for the new dataset.
2. We just trained OT for 30 epochs and uploaded the best model. Taking more steps with more reasonable adjustments to learning rates is encouraged so it is 100 in the code. We believe that the performance of the transformer-based model can be improved if more reasonable fine-tuning can be done.
3. We use this since we always want to construct the triplet loss containing both the positive and negative parts. It is OK to skip the triplet without positive samples.
4. We also trained OverlapNet for 30 epochs and found the best, which is similar to OT.

BIT-MJY, 2022-11-28:
Thanks for using our code again. I'm going to close this issue. If you have any questions, please feel free to reopen it.

closed 2022-11-28 by BIT-MJY