haomo-ai/OverlapTransformer #13: How to generate a three-channel depth map?

2653888596, 2022-11-16:
Hi, thank you very much for your work, and I admire it very much！
But I generated a single-channel depth map based on gen_depth_data in your utils. How can I generate a three-channel depth map with different colors like your paper? Attached is the depth map I generated.

[image]

BIT-MJY, 2022-11-16:
Hello @2653888596,

Thanks for your kind comments!

I guess this attached image is from your saved .png file.

You can directly run 

```
plt.imshow(your_img_arr)
plt.savefig("./test.png")
plt.show()
```
to visualize the colored depth map since this imshow function provides a default `cmap`.   You can also set this param such as  `plt.imshow(your_img_arr, cmap="viridis")`  to assign the specific color map.

2653888596, 2022-11-16:
@BIT-MJY Thank you very much!
Through your guidance, I have got the desired results. Thanks again!

BIT-MJY, 2022-11-16:
thanks for using our code again! I'm going to close this issue.

closed 2022-11-16 by BIT-MJY