haomo-ai/OverlapTransformer #14: How to generate the "train_set" file with my own lidar data?

jibowen69, 2023-02-06:
Hi, thank you very much for your work！
I would like to consult ”how to generate the "train_set" using my own data“，thank you very much.

BIT-MJY, 2023-02-08:
Hi @jibowen69 ,

Thanks for using our code! You can find the way to generate `train_set` in [demo 4](https://github.com/PRBonn/OverlapNet/blob/master/demo/demo4_gen_gt_files.py) of the OverlapNet repo.

jibowen69, 2023-02-12:
I'm sorry to bother you again, i want to consult how to the overlap data in the "gt_valid_folder", i use the  [demo 4] of the OverlapNet repo., but i only can get the "validation_set.npz", The data format of the data it contains is different from "gt_valid_folder/02/overlap_0.npy"

BIT-MJY, 2023-02-16:
You can generate `overlap_0.npy` by:
`
from utils.utils import *

def com_overlap_yaw(scan_paths, poses, frame_idx, leg_output_width=360):
    print('Start to compute ground truth overlap and yaw ...')
    overlaps = []
    current_points = load_vertex(scan_paths[frame_idx])
    current_range, project_points, _, _ = range_projection(current_points)
    visible_points = project_points[current_range > 0]
    valid_num = len(visible_points)
    current_pose = poses[frame_idx]

    for reference_idx in range(len(scan_paths)):
        reference_pose = poses[reference_idx]
        reference_points = load_vertex(scan_paths[reference_idx])
        reference_points_world = reference_pose.dot(reference_points.T).T
        reference_points_in_current = np.linalg.inv(current_pose).dot(reference_points_world.T).T
        reference_range, _, _, _ = range_projection(reference_points_in_current)

        overlap = np.count_nonzero(
            abs(reference_range[reference_range > 0] - current_range[reference_range > 0]) < 1) / valid_num
        overlaps.append(overlap)

    ground_truth_mapping = np.zeros((len(scan_paths), 3))
    ground_truth_mapping[:, 0] = np.ones(len(scan_paths)) * frame_idx
    ground_truth_mapping[:, 1] = np.arange(len(scan_paths))
    ground_truth_mapping[:, 2] = overlaps

    return ground_truth_mapping

if __name__ == '__main__':
    scan_folder = "/home/mjy/datasets/kitti/02/velodyne/sequences/02/velodyne"
    scan_paths = load_files(scan_folder)
    calib_file = ".../02/calib.txt"
    poses_file = ".../poses.txt"
    T_cam_velo = load_calib(calib_file)
    T_cam_velo = np.asarray(T_cam_velo).reshape((4, 4))
    T_velo_cam = np.linalg.inv(T_cam_velo)
    poses = load_poses(poses_file)
    pose0_inv = np.linalg.inv(poses[0])

    poses_new = []
    for pose in poses:
        poses_new.append(T_velo_cam.dot(pose0_inv).dot(pose).dot(T_cam_velo))
    poses = np.array(poses_new)

    frame_idx = 0
    ground_truth_mapping = com_overlap_yaw(scan_paths, poses, frame_idx=frame_idx, leg_output_width=360)
    np.save("./overlap_"+str(frame_idx), ground_truth_mapping)
`

jibowen69, 2023-02-19:
> 
Thanks for your help, it's useful! 
But i meet a new question, the `loop_gt_seq00_0.3overlap_inactive.npz` i generated by `[gen_gt_data.py]` is different with the file give, it result in the wrong PR value; Is this because i use "poses.txt" of kiiti instead of semantic kitti?
And i meet a error "_gen_gt_data.py:67: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray_" when turn list to array.

jibowen69, 2023-02-19:
the 'gen_gt_data.py' and 'loop_gt_seq00_0.3overlap_inactive.npz' i had update to https://github.com/jibowen69/ot_test.git

BIT-MJY, 2023-02-19:
> Is this because i use "poses.txt" of kiiti instead of semantic kitti?
Yes. You have to use semantic kitti poses which are more accurate. If you use the original kitti poses, the gt results are very different.

> And i meet a error "gen_gt_data.py:67
I think it's just a warning and may not affect the running.

jibowen69, 2023-02-19:
Thanks for your explanation!

closed 2023-02-19 by jibowen69