haomo-ai/OverlapTransformer #15: Recall@1% Ford Campus PR curve Questions.

2653888596, 2023-03-06:
Hello! I admire your work very much and appreciate your contribution to the field. After reading your paper and replicating your experiments, I have a few questions that I would be grateful if you could answer.

1.In Table 2 of your paper, there is a metric called Recall@1%. It is well-known that Recall@1 is equivalent to Top1 rate. What does Recall@1% mean in this context?

2.When replicating the experiments on the Ford Campus dataset, I extracted 3891 frames from the Ford dataset. However, the ground truth and pose.txt files provided on your website only have 3817 frames. Did you trim the dataset from the beginning (75-3891) or from the end (1-3817)? This information is crucial for the experiment, and I would appreciate it if you could clarify.

3.I have some questions regarding the PR curve. In your code, the faiss library is used for fast descriptor retrieval, and the resulting des_list is based on L2 distance. I'm not sure if my understanding is correct, but if it is, is it appropriate to directly compare this L2 distance to the threshold when calculating the PR curve?

Thank you very much for your help in addressing my concerns.

BIT-MJY, 2023-03-28:
Hi @2653888596 ,

Thanks for your interest in our work!

> 1.In Table 2 of your paper, there is a metric called Recall@1%. It is well-known that Recall@1 is equivalent to Top1 rate. What does Recall@1% mean in this context?

Please refer to this [issue](https://github.com/BIT-MJY/SeqOT/issues/10).

> 2.When replicating the experiments on the Ford Campus dataset, I extracted 3891 frames from the Ford dataset. However, the ground truth and pose.txt files provided on your website only have 3817 [...]

The raw dataset from the [official web](https://robots.engin.umich.edu/SoftwareData/Ford) contains 3817 frames. You can download it by 
```
wget  https://s3.us-east-2.amazonaws.com/ford.perl.engin.umich.edu/dataset-1.tar.gz
```

> 3.I have some questions regarding the PR curve. In your code, the faiss library is used for fast descriptor retrieval, and the resulting des_list is based on L2 distance. I'm not sure if my [...]

The similarity is strictly related to L2 distance. Therefore we can set thresholds for L2 distance to calculate PR curve. It is also ok to directly set thresholds for similarity. Just make sure that all the baselines have the same metrics for comparison.

2653888596, 2023-03-28:
Thank you very much for your answer. 
You have already helped me to answer question 1.
Question 2: The original data I downloaded from the official website was indeed 3817 frames. However, after analyzing the pcap package with the matlab code given by the author, I got 3891 frames.
Question 3: I have also communicated with our counterparts. As I have just entered this direction, thank you very much for your answer again!

closed 2023-03-28 by BIT-MJY