haomo-ai/OverlapTransformer #16: Some problems I'm having when trying to test my own dataset

shenyehui, 2023-04-25:
Thanks for your open source work! @BIT-MJY @Chen-Xieyuanli I plan to record a data set myself to test the effect of place recognition in the case of long-term or partial changes in the work, but in the process of looking at the code, I don't quite understand the generation of some files. 1. kitti experimental training uses the kitti03-09 data set, and the test uses the 00 data set, so what is the 02 data set in data_root used for? 2.  I plan to collect data for experiments according to the Haomo experiment, but in the haomo.ywl file, how are the three npy files triplets_for_training, pose_file_database and pose_flie_query generated? Could you please let us know if it is convenient for you?

shenyehui, 2023-04-27:
Why is it necessary to train the kitti datase with train_set.npz file, while the homo dataset does not require a similar file?

BIT-MJY, 2023-04-29:
1. seq 02 is used for fast validation during training. In the training file:
```
            with torch.no_grad():
                top1_rate = validate_seq_faiss(self.amodel, "02")
                writer1.add_scalar("top1_rate", top1_rate, global_step=i)
```

2. The pipeline of generating triplets_for_training follows the data processing for NCLT in [SeqOT repo](https://github.com/BIT-MJY/SeqOT#Data-Preparation). pose_file_database and pose_flie_query contain the array with the size of (n, 4, 4). You can simply produce the files by calculating the (4, 4) pose matrix for each pose node in the database and query trajectory.

> Why is it necessary to train the kitti datase with train_set.npz file, while the homo dataset does not require a similar file?
Because the data processing and metrics for KITTI and Haomo are different, leading to different data structures. Haomo contains similar and repetitive trajectories with some time gaps in the same environment, while KITTI has various trajectories in different environments.

shenyehui, 2023-05-05:
> Because the data processing and metrics for KITTI and Haomo are different, leading to different data structures. Haomo contains similar and repetitive trajectories with some time gaps in the same [...]
 Thank you for your answer！The different trajectories between the Haomo dataset and the Kitti dataset result in different data structures, so different data processing methods are used. Will this affect the performance of the model? Or can this indicate that the generalization performance of the model is not good enough?

BIT-MJY, 2023-05-05:
> Will this affect the performance of the model? Or can this indicate that the generalization performance of the model is not good enough?

For datasets with different data structures, different data processing methods are necessary. Besides, KITTI is used here for loop closing metrics and Haomo for place recognition metrics. Therefore, using different pre-processing codes has nothing to do with the generalization ability of our model. You can find the generalization study in Section V.B of our [paper](https://arxiv.org/pdf/2203.03397.pdf). Or you can refer to our newer work [SeqOT](https://github.com/BIT-MJY/SeqOT) for more details about evaluating generalization performance.

shenyehui, 2023-05-05:
> For datasets with different data structures, different data processing methods are necessary. Besides, KITTI is used here for loop closing metrics and Haomo for place recognition metrics. Therefore, [...]

Thank you very much. Your work has greatly inspired me, and I will always pay attention to and try to use your work.

BIT-MJY, 2023-05-05:
Thanks for following our work! We will release more open-source codes on place recognition in the future.

closed 2023-05-05 by BIT-MJY