haomo-ai/OverlapTransformer #18: Cannot get the claimed accuracy when running test demo

zouyajing, 2023-04-27:
Hi, thx a lot for your great effort.
I have followed the guidance and run the test demos, but the PR curve shows the recall@1 is about 0.64, while in the paper the value is 0.906. Below is what did to generate the PR curve:
[image]

lqf211, 2023-04-29:
I have also met this problem. Maybe you can try to use the script provided by OverlapNet to generate the depth files. I can achieve the results close to the paper using it instead of tools/utils/gen_depth_data.py. I think the difference is caused by the numerical precision of different codes.

BIT-MJY, 2023-04-30:
Hi @zouyajing,

Thanks for using our code! @lqf211 is right. We basically use the .png files provided by the author of OverlapNet for training and evaluation. We have released all the utilized [.png files](https://drive.google.com/file/d/1LOuhv9w7Oo-qK8SeGlpu2dxfIWIVta9T/view?usp=share_link) for testing in our experiments to simply reproduce the results in the paper. Besides, please modify the `max_range` for better performance if necessary.

zouyajing, 2023-04-30:
Hi @lqf211  and @BIT-MJY , thx for your help. I have downloaded the [kitti 00 range images](https://drive.google.com/file/d/1LOuhv9w7Oo-qK8SeGlpu2dxfIWIVta9T/view?usp=share_link). Also, the recall@1 I obtained is the same as the paper reported. But still the PR curve performance is not so good. Only recall is considered for comparison in the experiment, while precision is neglected, is that right? 
[image]

BIT-MJY, 2023-05-01:
> the recall@1 I obtained is the same as the paper reported. But still the PR curve performance is not so good.

Please use the same evaluation metric for all the baselines as I said before. Since OT uses the overlap-based metric to train and test, Iris (without shift matching) should also use the overlap-based metric for evaluation. Or you can use the distance-based metric for all the baselines. Only checking the shape of PR curves without considering metrics is not reasonable :)

BIT-MJY, 2023-05-05:
@zouyajing , we simply use a distance-base metric (3m positive) to evaluate OT trained on overlap-based kitti datasets. The PR results are attached here:
[image]
You can see that if you use different metrics, the results differ apparently.

This is a toy example. Actually, in order to ensure the fairness of the comparison, you'd better retrain OT using distance-based datasets as baselines did.

zouyajing, 2023-05-06:
@BIT-MJY. Thx a lot for your update. 
So the ground truth file of OT is generated by overlap, while that of Iris is by distance. If OT adopt distance-based ground truth to evaluate, it can yield better PR performance. Is that right?

BIT-MJY, 2023-05-06:
In theory yes, but we have not fully compared OT with baselines all trained on distance-based datasets. For overlap-based datasets, OT is the sota one as the paper posted.

zouyajing, 2023-05-07:
Yeah. I can understand that. In my early reply, the PR curve of OT is generated by comparing it with overlap-based ground truth, while that of Iris is by comparing it with distance-based ground truth. So this comparison is unfair. If compared with distance-based ground truth, OT can generate a better PR Break-Even Point.

Thx again for your answer. Furthermore, would you like to share the script to evaluate OT and generate PR curve using distance-based matrics. That will be very kind and helpful.

BIT-MJY, 2023-05-08:
Please follow our update on the new [issue](https://github.com/haomo-ai/OverlapTransformer/issues/20). I'm going to close this one :-)

closed 2023-05-08 by BIT-MJY