haomo-ai/OverlapTransformer #23: loss function question

yanlong95, 2023-06-27:
Hi, Junyi,

Thanks for sharing your excellent work. 

I have a question about your loss function implementation. In the paper, the loss function is kp*(**margin** + **max_p**(d(Vq, Vp))) - ..., but in the implementation, the loss function is **margin** + kp*(**max_n**(d(Vq, Vp))) - ... . Since the num_pos and num_neg are both 6 in the setup and the loss is fixed to be >= 0.0, seems it does not matter in the experiment. To confirm, if the change of loss function will affect the training result? 

Best,
Yanlong

BIT-MJY, 2023-06-30:
Hi @yanlong95 ,

Thanks for using our code! 

> kp*(**margin** + **max_p**(d(Vq, Vp))) - ..., but in the implementation, the loss function is **margin** + kp*(**max_n**(d(Vq, Vp))) - ... 

They are basically the same since you can regard **kp*margin** as **margin**. The margin matters because although the loss is fixed to be >= 0.0, we can still train the samples with negative loss less than margin. 

`    loss = margin + positive - ((neg_vecs - query_copies) ** 2).sum(1).unsqueeze(1)
    loss = loss.clamp(min=0.0)
`

Therefore, the change of loss function, such as removing the margin, does affect the training result.

yanlong95, 2023-06-30:
Thanks a lot for your explanation.

closed 2023-06-30 by yanlong95