haomo-ai/OverlapTransformer #25: Requests for related files about Haomo dataset.

cpkingw, 2023-07-05:
Hi, i am interested in the results of related work on the Haomo dataset. Do you have a plan to release the related files about Haomo datasets like KITTI?

BIT-MJY, 2023-07-10:
Hi @cpkingw , Thanks for your interest in our work. We have released the related data [here](https://github.com/haomo-ai/OverlapTransformer/tree/master/Haomo_Dataset), but the server migration leads to the deprecated links. We will update the links upon the migration is done. Please refer to this [issue](https://github.com/haomo-ai/OverlapTransformer/issues/24).

closed 2023-07-10 by BIT-MJY

cpkingw, 2023-07-12:
Could you release the corresponding groundtruth and trainning files of Haomo Dataset?