haomo-ai/OverlapTransformer #29: 自己的数据集的train_set.npz如何制作

yxxzhuzhuxia, 2024-04-07:
首先，感谢您优秀的工作，我对train_set.npz有一些疑惑，我用脚本简单查看了您提供的02段train_set.npz的数据如下
[image]
我用overlapnet使用demo4_gen_gt_files.py生成的数据如下
[image]
我理解的是overlapnet的demo4脚本生成的第一列是第零帧，第二列其他帧，第三列是重叠度，第四列是yaw角，我想修改程序生成相应的(57413, 3)的train_set.npz，但是我不太理解您提供的数据，我理解第一列和第二列是帧索引，第三列是两帧的重叠度，请问是这样子的意思吗？

BIT-MJY, 2024-04-08:
@yxxzhuzhuxia 感谢关注我们的工作。

> 第一列和第二列是帧索引，第三列是两帧的重叠度

这个理解是正确的。最后生成的训练集是根据overlap进行normalize过的。

yxxzhuzhuxia, 2024-04-08:
感谢您的回答

closed 2024-04-08 by yxxzhuzhuxia