haomo-ai/OverlapTransformer #31: does it works  only for range images ( horizontal 360 degree depth )

ynma-hanvo, 2024-05-22:
hi,  thanks for the great contribution on loop closure detection work.  
I am wondering does it apply to   sensor  like rgbd camera (which   have fov  < 180 degress )

BIT-MJY, 2024-05-22:
@ynma-hanvo ，Thanks for your interest in our work.

OverlapTransformer's inherent yaw-angle invariance would not hold once the FOV is not 360 degrees. This may lead to significant performance drop because OT is not a sophisticated model but holds natural mathematical property in a lightweight style. We have not used sensors with limited FOV such as RGBD camera providing its inputs. You can try it and welcome to post results here :)

Btw, our enhanced version [CVTNet](https://github.com/BIT-MJY/CVTNet) shows good performance on solid state LiDAR, which may be helpful to your work.

ynma-hanvo, 2024-05-22:
> @ynma-hanvo ，Thanks for your interest in our work.
> 
> OverlapTransformer's inherent yaw-angle invariance would not hold once the FOV is not 360 degrees. This may lead to significant performance drop because OT is not a sophisticated model but holds natural mathematical property in a lightweight style. We have not used sensors with limited FOV such as RGBD camera providing its inputs. You can try it and welcome to post results here :)
> 
> Btw, our enhanced version [CVTNet](https://github.com/BIT-MJY/CVTNet) shows good performance on solid state LiDAR, which is somehow more robust than OT in FOV and may be helpful to your work.

thanks. this issue is closed as the reply answers me very well.

closed 2024-05-22 by ynma-hanvo

ynma-hanvo, 2024-05-24:
hi,  I tested with the CVTNet,  found it is possible to detect  similar view even with  narrow fov.   
but for another simple scene with  simple structures around,  it also reported  quite a lot of false positive. 
I   used the descriptor distance  with < 0.1 as   a positive match.

BIT-MJY, 2024-05-24:
@ynma-hanvo Thanks for reporting the results! Did you use it in indoor environments? We have not conducted any indoor experiments since it was originally designed for large-scale outdoor environments. Could you please provide more details?

ynma-hanvo, 2024-05-27:
> @ynma-hanvo Thanks for reporting the results! Did you use it in indoor environments? We have not conducted any indoor experiments since it was originally designed for large-scale outdoor environments. Could you please provide more details?
 
Hi, I used only a rgbd camera with max distance clamped to 5m for the indoor scenes.  general observation is that it detects   simliar pointclouds quite well.  But as the indoor scene is  simple and due to the narrow fov,  false positive is not avoidable.  I think  It would be more accurate   with more features like pixel color  added  .